tedyapo/arduino-MLX90393 #9: Reduce code size

tedyapo, 2016-07-24:
as written the code takes up a lot of flash on the ardunio.  Find ways to reduce footprint

udoklein, 2018-01-15:
I assume the biggest size reduction would be to get rid of floating point arithmetic.

if I look at

  gain_multipliers[0] = 5.f;
  gain_multipliers[1] = 4.f;
  gain_multipliers[2] = 3.f;
  gain_multipliers[3] = 2.5f;
  gain_multipliers[4] = 2.f;
  gain_multipliers[5] = 1.66666667f;
  gain_multipliers[6] = 1.33333333f;
  gain_multipliers[7] = 1.f;

these are fractions with a lowest common divisor of 6. Thus I suggest to use the following uint8_t gain multipliers:

  gain_multipliers[0] = 30
  gain_multipliers[1] = 24
  gain_multipliers[2] = 18
  gain_multipliers[3] = 15
  gain_multipliers[4] = 12
  gain_multipliers[5] = 8
  gain_multipliers[6] = 7
  gain_multipliers[8] = 6

I also suggest to change the floating point arithmetics to int32_t. Thus the library would only use integer arithmetic. Also all consumers would benefit from it by not linking in floating point code.

If someone definitely requires to have it in floats division by 6 is something that can be explained in the documentation. For my current project this would be a size reduction without any drawbacks. The reason being that I can not do any calibration anyway. Hence I am looking only at relative values. Thus a scale factor is completely irrelevant. I assume that most projects lack the equipment to calibrate. Hence floating point values would be somewhat pointless for most.

What do you think?

udoklein, 2018-01-15:
This would also reduce SRAM use as well.

tedyapo, 2018-01-15:
Yes, I agree, removing floating point arithmetic probably has the most potential to relieve size (and speed) issues for smaller 8-bit MCU's.  I have a few reservations, though.

1.  I know the 8-bit arduino lives on, but the availability of much more powerful arduino-programmed boards (Teensy and ESP32 come to mind) make the size and speed issues less important.  These boards are around the same price as an 8-bit arduino (from a reputable source), and have vastly more resources.  For one-off development and prototyping, I would suggest people take advantage of these higher-powered boards.

2. There are more than the gain multipliers involved.  There are also the four base sensitivities:

  // for hallconf = 0
  base_xy_sens_hc0 = 0.196f;
  base_z_sens_hc0 = 0.316f;
  // for hallconf = 0xc
  base_xy_sens_hc0xc = 0.150f;
  base_z_sens_hc0xc = 0.242f;

3. The MLX90393 is calibrated at the factory, and with the coefficients from the datasheet, isn't really all that bad.  It would be a shame to give up the factory calibration.

One possibility is to go to 32-bit fixed-point (as opposed to integer) arithmetic.  That could preserve enough precision while speeding execution and removing floating point libraries.  It's a lot of work and even more testing, though.

My other thought is that for many applications, you probably only need a subset of the commands and configuration of the MLX90393.  In those cases, if you are really size-constrained, you could use the part directly, as is done here:

https://github.com/ControlEverythingCommunity/MLX90393/blob/master/Arduino/MLX90393.ino

You could, for instance, develop your application on an ESP32 using the floating-point code, then move to your target 8-bit platform with a stripped-down interface like the linked one.

udoklein, 2018-01-15:
It would be possible to split the class into two classes / libraries. One for the bare bones access without floating point conversion. The other one with the conversion. Thus depending on what you use only the integer part would be linked into the compiled program. 

I agree with the point that there are other more powerful boards around.

The issue is which audience is the library targeting. Beginners will never get this point and stick to Arduino.

tedyapo, 2018-01-17:
If you are suggesting a fork for the two classes/libraries, I think that's probably right.  A lightweight integer/raw data version, and one with nice, pretty floats.

Another thing to eliminate is the state caching.  There is a lot of code devoted to maintaining the state of the MLX90393 on the MCU so that we don't have to query over I2C for parameters needed for each conversion.  You could really reduce code size by just asking the part for the registers/fields as required.  It would be slower, but reduce code size.  I'm not sure where to make that trade-off.

Ansar424, 2018-07-11:
Dear Tedyapo,

How Can I impliment library files or driver files for MLX90393 that works with XILINX SDK for my project of a sensor ring of 13 MLX90393 that can controlled by XILINX

tedyapo, 2018-07-12:
Without more information about your project, it's difficult to say other than "just do it".  I would guess porting the existing code to whatever microcontroller IP core you've chosen would be a good start.

Also, this question isn't related to the Issue you posted it under, since it has nothing to do with reducing the code size.  Please open new issues for new things.

closed 2018-09-22 by tedyapo