tedyapo/arduino-MLX90393 #11: .begin() always returns 255

nseidle, 2017-04-20:
Hi Tedy - thanks for writing this lib.

I think I've found a bug:

During a .begin, a reset() is issued. Immediately after, setGain() is issued. setGain fails ~80% of the time (returns 255) because the lib immediately tries to read the first memory location

    uint8_t status1 = readRegister(GAIN_SEL_REG, old_val);

and it can't because the IC is still resetting. This causes the .begin to return 255 (error).

POR is 1.6ms max. Software reset time limit is not specified.  We could poll register until it reads correctly, but I did a bit of a hack in the reset() function:

```
uint8_t
MLX90393::
reset()
{
  invalidateCache();
  uint8_t cmd = CMD_RESET;
  
  sendCommand(cmd); //Device now resets. We must give it time to complete
  delay(2); //POR is 1.6ms max. Software reset time limit is not specified. 2ms was found to be good.
  return sendCommand(cmd);
}
```
This works well for me. I'll try to issue a pull request in a moment.

tedyapo, 2017-04-21:
Yes, my pleasure.  Thanks for your interest and enhancements, and, you know ... Sparkfun.

I commented in the merge about either exposing the delay as a defaulted parameter or doing polling as you suggest, but for the time being, the 2ms delay seems fine.

closed 2017-04-21 by tedyapo

nseidle, 2017-05-08:
SparkFun = no problem :) Thanks!

Passing delay by argument with default is a good solution. I'll try to do that the next time I'm in the lib.