tedyapo/arduino-MLX90393 #16: reset does not return error status

udoklein, 2018-01-09:
Currently reset will always return error code OK. In my opinion it is possible that the command fails. I suggest to return the status from sendCommand instead of discarding it.

tedyapo, 2018-01-09:
Thanks for the reports!  It don't have a testbed set up for regression tests at the moment.  It will probably take a few days to find time to hook up a part so I can verify the changes before merging them in.

tedyapo, 2018-01-09:
I pushed the api_cleanup branch which I think has all your recommended changes.  It may take a little while until I can test it then merge it in. Until then, you can use that branch if you like.

Thanks for your help!

udoklein, 2018-01-09:
I am currently writing a "debug helper" testbed such that I better understand how the library and the chip work. Once I am done I can submit it. So far my understanding is already going up significantly be playing around with the chip.

One thing that struck me is that it appears that I can not read registers from the chip once it is in batch mode. I wonder if this interferes with the cache logic. Do you have any opinion on that? If not: I will find out soon once my debug helper gets more functions. I will then report what I find.

tedyapo, 2018-01-09:
It's certainly possible that batch mode prevents you from reading
registers; I don't know that I've tried.

Unfortunately, Melexis isn't of much help, either.  I've contacted them
before and they don't have time for small players.

To learn more, some hacking and experimentation is probably required.

Thanks for reviewing the code and reporting these issues!

udoklein, 2018-01-09:
Well, thanks for providing the library in the first place :)

tedyapo, 2018-01-11:
fixed in #28

closed 2018-01-11 by tedyapo

gituuu, 2020-04-19:
In the latest version of the datasheet it says:

"The MLX90393 only listens to a specific set of commands. Apart from the Reset command, all commands generate a status byte that can be read out."

gituuu, 2020-04-19:
Reset is messing up my I2C bus and i have to perform a dummy write after doing a reset, otherwise I get an error 121 (on Linux).

tedyapo, 2020-08-25:
do you believe that inappropriately reading the status of the reset command is causing your issue?