tedyapo/arduino-MLX90393 #20: setTemperatureOversampling does not work as expected

udoklein, 2018-01-09:
This one is strange. If I call setTemperatureOversampling and read the register back then it shows that OSR2 equals 0. I added some print statement into the writeRegister function. It appears to me that it is called properly. Reviewing your code I would says the code is correct. This is consistent with the runtime value found by my print statement.

Since it works as expected for the other cases I am inclined to suspect that either
1) the Melexis 90393 does not behave as expected or
2) my instance of the Melexis 90393 does not behave properly.

Did you ever test this function? Did it work for you? Since I do not intend to read temperature at all this is not big deal. So if you have no idea why this does not work do not bother with this one. I will ignore it then as well.

tedyapo, 2018-01-09:
I didn't write formal test code for all the functions, so I can't verify
that I ever tested reading that value back.

Once I get set up here, I can verify with some additional samples of the
part.  If it's a chip bug, we should report it to Melexis.

udoklein, 2018-01-10:
By now I found it. The error is in this line:

 enum { OSR2_REG = 0x2, OSR2_MASK = 0x0180, OSR2_SHIFT = 11 };

The mask should be 0x3<<11 which is 0x1800 instead of 0x0180.

tedyapo, 2018-01-10:
Nice find! So, that's what a bug looks like?  Never seen one before :-)
Fixed in api_cleanup

tedyapo, 2018-01-11:
#28

closed 2018-01-11 by tedyapo