tedyapo/arduino-MLX90393 #21: gain_multipliers[] has 7 elements but 8 values are accessed

udoklein, 2018-01-10:
The line 

float gain_factor = gain_multipliers[7 - (gain_sel & 0x7)];

and the fact that the datasheet says that gain_sel ranges from 0-7 make me wonder what convertRaw will do in case of gain_sel == 0. I also wonder why you index with 7- gain_sel instead of sorting the array the other ways round.

My conjecture is that the factor "1.0" is missing in the array.

udoklein, 2018-01-10:
If it is OK, please do not fix yet. I will submit a not so obvious patch for this.

udoklein, 2018-01-10:
Oops, forget my previous comment. I played a little bit with progmem. It did decrease the size of the sketch but increased the pressure on SRAM. I do not understand why.

I suggest to just add the missing factor 1.f for gain_sel == 0. I also recommend to revert the array and thus make the lookup slightly simpler.

tedyapo, 2018-01-10:
Yes, there should be 8 values, which is a bug, but there is much more going on here.
I'm looking at the latest version of the MLX90393 datasheet, and it's different from the one I wrote the driver from (I saved a copy of the previous version).  The scaling values are different, and the new document describes different scaling based on the value of HALLCONF, which was previously undocumented.

Based on this, the whole scaling code needs to be re-written to reflect the values in the new version of the datasheet.  I suspect the whole driver should be reviewed in light of these changes.

I do not know yet if the new version of the datasheet applies to all versions of the silicon, or if changes have been made to the part.

tedyapo, 2018-01-10:
The new datasheet would explain the strange scaling factors I determined from Helmholtz coils, [documented here] (https://hackaday.io/project/11865-3d-magnetic-field-scanner/log/40164-first-mlx90393-tests).  I wrote at that time that I suspected the scaling factors in the datasheet could be wrong.  The new datasheet seems to correct for the 1.27x factor I measured.
I need to drag my Hemlholtz coil out of storage and verify once we get the driver straightened out.

commit 46dac52a (2018-01-11): fixed #21 - missing factor 1 for gain_sel == 7
commit 71ba5a32 (2018-01-11): Merge pull request #22 from udoklein/master

fixed #21 - missing factor 1 for gain_sel == 7
closed 2018-01-11 by tedyapo