tedyapo/arduino-MLX90393 #30: esp32

jschoch, 2018-01-13:
I'm unable to get the sample code working with my wemos lolin 32 board.  The values do not update.  here is the sketch

```
#include <Wire.h>
#include <Arduino.h>
#include <MLX90393.h>

// prints Bx By Bz (in uT) and temperature (C) to serial console after receiving a character

MLX90393 mlx;

void setup(){
  // DRDY line connected to A3 (omit third parameter to used timed reads)
  //uint8_t status = mlx.begin(0, 0, A3);
  
  // this seems to un-funk other i2c stuff for esp32.  My board has SDA and SCL 
  //  mapped to these pins.
  Wire.begin(5,4);

  delay(300);
  uint8_t status = mlx.begin(0, 0);
  //uint8_t status = mlx.begin();
  
  Serial.begin(115200);
  Serial.println("setup done");
}

void loop(){
  MLX90393::txyz data;
  if (Serial.available()){
    Serial.read();
    mlx.readData(data);
    Serial.print(data.x);
    Serial.print(" ");
    Serial.print(data.y);
    Serial.print(" ");
    Serial.print(data.z);
    Serial.print(" ");
    Serial.println(data.t);
  delay(100);
  }
}
```

following is a sample of the output.  oddly a few of the values were in fact changed.

```
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c
setup done
-25679.13 -17812.48 -82857.73 -964.12
-0.78 4224.98 -82797.06 446.15
-25683.84 -17825.02 -82857.73 -907.41
-25679.13 -17812.48 -82857.73 -964.12
-15404.82 0.00 -165675.02 446.22
-25683.84 -17825.02 -82857.73 -907.41
-25679.13 -17812.48 -82857.73 -964.12
0.00 0.00 -165675.02 -998.03
0.00 0.00 -165675.02 -998.03
-25679.13 -17812.48 -82857.73 -964.12
-25679.13 -17812.48 -82857.73 -964.12
-25679.13 -17812.48 -82857.73 -964.12
-25683.84 -17825.02 -82857.73 -907.41
-25679.13 -17812.48 -82857.73 -964.12
-25683.84 -17825.02 -82857.73 -907.41
-25679.13 -17812.48 -82857.73 -964.12
-25679.13 -17812.48 -82857.73 -964.12
-25679.13 -17812.48 -82857.73 -964.05
-25683.84 -17825.02 -82857.73 -907.41
-25679.13 -17812.48 -82857.73 -964.05
-25683.84 -17825.02 -82857.73 -907.41
-0.78 4224.98 -82797.06 446.15
-25679.13 -17812.48 -82857.73 -964.12
-25683.84 -17825.02 -82857.73 -907.41
-25683.84 -17825.02 -82857.73 -907.41
-25679.13 -17812.48 -82857.73 -964.12
-25683.84 -17825.02 -82857.73 -907.41
-25679.13 -17812.48 -82857.73 -964.12
-25679.13 -17812.48 -82857.73 -964.12

```

the following does work (but not great):

https://gist.github.com/jschoch/779e7aae5205bfbcb93a064da627201e

source for this is: 
https://github.com/ControlEverythingCommunity/MLX90393/blob/master/Arduino/MLX90393.ino

tedyapo, 2018-01-13:
What is the I2C address of the OLED diplay on your I2C bus?

jschoch, 2018-01-13:
should be 0x3c

jschoch, 2018-01-13:
here is the output of an i2c scanner when the device is hooked up.  You can see both the OLED screen and the mlx device at the correct addresses.

```

start scan
 Scanning I2C Addresses
.. .. .. .. .. .. .. .. .. .. .. .. 0C .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. 3C .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
Scan Completed, 2 I2C Devices found.

Discovering eeprom sizes 0x50..0x57
0x50: Not Present.
0x51: Not Present.
0x52: Not Present.
0x53: Not Present.
0x54: Not Present.
0x55: Not Present.
0x56: Not Present.
0x57: Not Present.
done setup

```

tedyapo, 2018-01-13:
One more quick test: dump the value returned by readData():
uint8_t status;
status = mlx.readData(data);
Serial.print("status = ");
Serial.println(status);
see if an error code is returned

jschoch, 2018-01-13:
i recall it was 255 previously.  I'll test it again.

jschoch, 2018-01-13:
confirmed status = 255.

jschoch, 2018-01-13:
also hooked up an uno (my pro micro is 5v without a 3.3v) to a level shifter and have a sensor working with status 0.

tedyapo, 2018-01-13:
I have tested it with an [ESP32 dev board](https://www.amazon.com/HiLetgo-ESP-WROOM-32-Development-Microcontroller-Integrated/dp/B0718T232Z) here, and it works as expected.  So, unfortunately, I am unable to reproduce the problem at this time.

I ordered one of those OLED ESP32 boards, so when that comes I can try again.

You can check the status of the begin() call to see if it fails that early.  Next steps are printing values within the begin() call to find the first one that fails.

What MLX90393 PCB are you using?

Interestingly, I first tested the code on an ESP32 without the sensor connected at all, and received similar garbage values as you posted above.  I'd suspect that the I2C connections were incorrect, but the scanner finds it on the bus.

tedyapo, 2018-01-13:
One other test is to try using another pair of pins on the ESP32 for I2C.  You won't be able to use the display, but it might yield some clues.

tedyapo, 2018-01-13:
Oh, also, I did not need to modify the original sketch from the repo.  The ESP32 dev board has SCL on pin 22 and SDA on pin 21.  Connecting the sensor to those works for me.

jschoch, 2018-01-14:
I have a doit esp32 module, but I was not able to get Wire.begin(23,22) working.  I will try a few other pins. 

Odd the other code referenced above works Wemos with the OLED.  Maybe I should try with the SDA and SLC pulled up to 3.3v with a 2.4k resister

tedyapo, 2018-01-14:
I found it.  The library breaks if you call Wire.begin() before mlx.begin().  Working on a fix now, but you can verify by commenting out line 63 of MLX90393.cpp:

//_i2cPort->begin();

tedyapo, 2018-01-14:
fixed in [fix_wire_begin branch](https://github.com/tedyapo/arduino-MLX90393/commit/f5c50366575a13a3c41f98783e9ee1eae580829e).

You always have to call either Wire.begin() or Wire.begin(sda, scl) now, whether you need to change the pins or not.

Please verify this works for you, and I will close the issue and merge into master.

jschoch, 2018-01-14:
sweet!  working on my board!  Thank for the help!!!

tedyapo, 2018-01-14:
Thank you for finding this and chasing it down!

This issue has been there for a while.  I'm refraining from hitting the "blame" button on the master branch :-)

Merged into master #31.

closed 2018-01-14 by tedyapo