tedyapo/arduino-MLX90393 #32: interrupt question

jschoch, 2018-01-17:
wondering if you are polling the interrupt pin for a reason, or if is more of a placeholder function?  I'd like to map it to a hardware interrupt on the esp32 (or STM32 which I havn't tried yet.

BTW i bought 2 of these from ali express for $6 each.  the part is CJMCU-90393.  I think you had asked in a previous post.

tedyapo, 2018-01-17:
I just poll the interrupt line to know when the sample is ready as opposed to delaying for a fixed time.  My thought is that in "arduino land," if you are advanced enough to write an interrupt-driven system, you are best off writing it yourself.  In any system where using interrupts from this part matters, the coupling between interrupts and the rest of the code is probably complex enough that what I could support in a general-purpose library probably isn't general-purpose anymore.

There is also the issue of communication between the call that initiates the conversion and the subsequent data read.  What you get back from the IC depends on what you asked, so you need to keep track of that to read the data correctly, anyway.

My suggestion is to use the lower-level interface to implement an interrupt-driven system:

1. Call
        startMeasurement(uint8_t zyxt_flags)
    to begin the measurement
2. When you are interrupted because the conversion is done, call
       readMeasurement(uint8_t zyxt_flags, txyzRaw& txyz_result);
    to read the data back from the part
3. Call 
       convertRaw(txyzRaw raw);
    to convert the raw measurements to microteslas

I see now that convertRaw() is private (as is checkStatus()).  I'll fix that now.

Does this accomplish what you seek?

tedyapo, 2018-01-17:
convertRaw() and checkStatus() are now public.

jschoch, 2018-01-18:
Thanks!!!  Unfortunately I"m having stability problems which I think are related to the many problems folks have had with the esp32's i2c bus.  It runs ok for a minute or two. and then starts to throw the 255 status after a progressive slowdown.  I just moved my setup over to an esp8266 and it seems very stable.  I'll try to test again when I have time to dig into the esp32 again.  I can test the changes above on the 8266 tomorrow I think.

closed 2018-01-18 by jschoch