tedyapo/arduino-MLX90393 #34: Question: how did you get access to the 2017 datasheet?

udoklein, 2018-01-17:
I noticed that my datasheet was from 2015. So I tried to download the 2017 datasheet from the Melexis website. However it appears to be encrypted with a password. Do you know this password? Or do you know where to get an unencrypted version of the file?

tedyapo, 2018-01-17:
LOL!

I went back and forth in email two weeks ago with a Melexis sales rep insisting that the datasheet was encrypted and I needed a password, but they didn't believe me.  The rep suggested I use a work email account instead of gmail (?). I think the PDF is just using a very modern (and maybe incorrect) encryption that is supposed to let you read the file, but not copy from it.

Most Linux readers won't open it (including the command-line pdfinfo).  I have also had difficulty opening it on a Win10 machine.  I asked a bunch of people to try, and some were able to open it with more leading-edge linux distros which have very up-to-date PDF support.

I finally found the [foxit pdf reader](https://www.foxitsoftware.com/pdf-reader/) which runs on my linux box and will read the file.

My suggestion is to complain to Melexis like I did.  Maybe if enough people complain, they will finally understand.

tedyapo, 2018-01-17:
Oh, you can also open with the built-in PDF reader in Firefox (57.0.4).  Mexlexis' web site is mis-configured so by default the PDF won't open in the browser, but if you save the file, then use "ctrl-o" to open it from disk, it works.

closed 2018-01-17 by tedyapo

udoklein, 2018-01-17:
Thanks for this hint. The wonders of modern content protection.

udoklein, 2018-01-18:
With your hint I now figured out how to most easily read this under Linux. Use qpdf to figure out that the encryption is encrypted with an empty password. 

```
> qpdf *2017.pdf --show-encryption
R = 6
P = -1340
User password = 
extract for accessibility: allowed
extract for any purpose: not allowed
print low resolution: allowed
print high resolution: allowed
modify document assembly: not allowed
modify forms: not allowed
modify annotations: not allowed
modify other: not allowed
modify anything: not allowed
stream encryption method: AESv3
string encryption method: AESv3
file encryption method: AESv3
> qpdf *2017.pdf decrypted.pdf --decrypt
>
```

Decrypt it using qpdf without telling it a password. I am deeply impressed by this protection scheme ;)

tedyapo, 2018-01-18:
Nice!

I think was most likely a mistake, either the software they used to prepare
the document is buggy, or someone actually typed a blank password.  Ether
way, some pdf readers seem to just ignore (or perhaps supply) the empty
password.