tedyapo/arduino-MLX90393 #38: mlx.begin() crashes my arduino

joshjohner, 2018-05-21:
My MLX is a MJMCU-90303
Board is a Elegoo Uno R3 and a Teensy 3.6 (tried both)

I am an absolute novice at anything and everything programming and microcontrollers so I may be missing something extremely simple. I am using the example code from sparkfun.com 

https://www.sparkfun.com/products/retired/14160

In the sketch, as soon as it calls the mlx.begin() function it hangs. I have inserted Serial.print() messages between each line of code to isolate what is causing the problem and verified it is the mlx.begin(). Strangely, it will even stop the arduino from printing the "MLX90393 Test" message that takes place 6 lines prior. It will print out "ML" and then nothing else. If I add a delay(500) prior to the mlx.begin() call, it will print everything prior to the call in the serial monitor. I have also tried adding a delay(1000) after the mlx.begin() to see if it was just having a tough time trying to print something while the begin function was still trying to finish. Has anyone else had this issue or know what may be causing it?

joshjohner, 2018-05-21:
I forgot to mention that I was able to find another sketch that uses only the wire library to verify that the breakout is functioning communicating properly.

tedyapo, 2018-05-21:
The code supplied on the sparkfun site is written for an old version of the
library.  Sparkfun has a fork of the library (33 commits behind on master)
that may work with their example code.  You can see it at:

https://github.com/sparkfunX/arduino-MLX90393

Sparkfun's fork of the library and their code may be consistent.  I don't
know - that would be a question for them.

If you want to use the newest library, you have to call either Wire.begin()
or Wire.begin(sda, scl) before calling mlx.begin().  This change was added
so that you could use alternate I2C ports on boards which support it.

Please add a Wire.begin() call before mlx.begin() an see if that solves
your issue.

joshjohner, 2018-05-21:
That fixed it! Well, sort of. Now I am getting a constant value for the fields regardless what I do with my magnet. Also, the temp is 305C. Do you know of any good tutorials or documention/examples that are up to date for me to learn?

tedyapo, 2018-05-21:
You should check the return values of the calls into the library - printing
them out is fine. I noticed that the example code you referenced isn't
checking the values. I suspect you will find one or more of them returning
an error code indicating something is wrong.

joshjohner, 2018-05-22:
I apologize for taking so long to get back to you. I have been learning classes/structs/header files as I go so this has been a slow process.

Here is my current code

`#include <Wire.h>
#include <MLX90393.h>

MLX90393 mlx;
MLX90303::txyz data;

void setup()
{
  Serial.begin(9600);
  Serial.println("MLX90393 Read Example");
  
  //Connect to sensor with I2C address jumpers set: A1 = 1, A0 = 0
  //Use DRDY pin connected to A3
  //Returns byte containing status bytes
  Wire.begin();
  byte status = mlx.begin();

  //Report status from configuration
  Serial.print("Start status: 0x");
  if(status < 0x10) Serial.print("0"); //Pretty output
  Serial.println(status, HEX);    
}

void loop()
{
  Serial.print("Start: ");
  byte error = mlx.startMeasurement(0x2 | 0x4 | 0x8 | 0x1);
  error = mlx.checkStatus(error);
  Serial.print("0x");
  Serial.println(error, HEX);
  Serial.print("Read: ");
  MLX90393::txyzRaw raw_txyz;
  error = mlx.readMeasurement(0x2 | 0x4 | 0x8 | 0x1, raw_txyz);
  error = mlx.checkStatus(error);
  Serial.print("0x");
  Serial.println(error);

delay (1);
}
`

My output is 

> MLX90393 Read Example
Start status: 0x00
Start: 0x0
Read: OxFF
Start: 0xFF
Read: OxFF
Start: 0xFF
Read: OxFF
Start: 0x0
Read: OxFF

I do notice when I run the example as it was provided (to get the xyz converted readings) every few loops I will get what appear to be reliable numbers.

tedyapo, 2018-05-22:
If you use the low-level API (including the startMeasurement() and
readMeasurement() calls), you are responsible for making sure the
conversion has completed before you try to read the results. Your code
lacks an appropriate delay or polling or interrupt to know when the
conversion has completed.

I would start with the higher-level API. Calling readData() takes care of
all of this for you.

joshjohner, 2018-05-23:
It has randomly started working. For a few passes, it would work once, then when I uploaded the exact code again, it would not work, then after a couple uploads it would. I'm thinking my board may have been wired wrong a few too many times in the recent past and is internally damaged. I switched back over to my Teensy (newer) and it seems to be working normally now. 

Are there any docs or tutorials I can find anywhere on how to calibrate this thing? Also to convert my x,y,z output into x,y for a joystick application?

tedyapo, 2018-05-29:
I would check the Melexis application notes for clues on how to use this sensor in a joystick application

closed 2018-05-29 by tedyapo