tedyapo/arduino-MLX90393 #40: weird distortion at corners - joystick application

git-tiger, 2018-09-04:
Trying to design a joystick using the MLX90393.
Using the code in this library, I converted core code to Python/raspberry pi.
Communicating with the device using SPI, I can get a x/y angle of +/-30degrees from joystick tilt.
Problem is with the corners, the angle gets reported larger than 30degrees, which creates spikes outwards.
Does anyone know why this happens?

git-tiger, 2018-09-04:
Image of the corner issue
[https://drive.google.com/open?id=10TBIbMajsiFhMsbUNz5z_pK-BCS_CI4a](url)

tedyapo, 2018-09-19:
Hmm.  You'll have to provide a little more info.  What does your joystick design look like, and why do you think the resulting data should be square?

git-tiger, 2018-09-20:
[image]

tedyapo, 2018-09-22:
I wrote a quick simulation of your setup, and I'm pretty sure this distortion comes from the curvature of the field around the disk magnet.  You can see a video of the simulation here:

[magnet joystick simulation video](https://www.youtube.com/watch?v=8i8qMIaZ0h8&feature=youtu.be)

Here's a single frame.  The red and blue dots represent a collection of magnetic dipoles to simulate the disk magnet.  The sensor is at the black dot on top.

[image]

You can see the curved field away from the axis of the magnet.  That seems to be the problem.  If I re-run the simulation with a uniform magnetic field instead, the distortion goes away:

[joystick with uniform field video](https://www.youtube.com/watch?v=xJ_JFciQXkI&feature=youtu.be)

[image]

Two things come to mind.  First, this distortion looks very similar to pincushion distortion caused by some lens systems.  See the [wikipedia article](https://en.wikipedia.org/wiki/Distortion_(optics)#Radial_distortion) for more info.  You may be able to adequately correct the response using a model for the radial distortion as explained in the article.  You could either choose coefficients by trial and error, or use your collected data to calibrate the joystick.

Also, if you are willing to change the design, and you really need a linear response, you could use a Helholtz (or Maxwell) coil to generate a uniform field.  In this case, I would probably mount the sensor on the joystick and keep the field stationary.  This solution would consume more power, since you'd need to energize the coil to create the field, but should eliminate most of the distortion.

I wrote the simulation in python.  I'll probably clean it up and release it as a more general package over time, but let me know if you want to play with it sooner.

tedyapo, 2018-09-22:
Update: it's not because the field lines are curved, it's because of the way the extended geometry of the disk magnet curves the field lines.  If you use a single, tiny dipole as the magnet, the output is linear, even though the field lines are all circles.  You can show this on paper fairly easily for the field from a dipole.

I think the answer is that for any actual magnet, you'll need to calibrate the joystick to produce a linear output, if that's what you need.

closed 2018-09-22 by tedyapo