tedyapo/arduino-MLX90393 #44: Issue with other library

Jim-bee, 2019-07-15:
Hi,
Your library works great.  I do have one issue when I include sumotoy ILI9163C library there is a conflict with both libraries.  Both libraries work fine on their own.

Please let me know if this can be fixed or if there is a work around.

Compile Error;

n file included from /Users/JimD/Documents/Arduino/libraries/TFT_ILI9163C-master/TFT_ILI9163C.h:135:0,
                 from /Users/JimD/Documents/Arduino/RewinderTS/RewinderTS.ino:11:
/Users/JimD/Documents/Arduino/libraries/TFT_ILI9163C-master/_settings/TFT_ILI9163C_registers.h:6:22: error: expected identifier before numeric constant
 #define CMD_NOP      0x00//Non operation
                      ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:38:5: note: in expansion of macro 'CMD_NOP'
     CMD_NOP = 0x00,
     ^
/Users/JimD/Documents/Arduino/libraries/TFT_ILI9163C-master/_settings/TFT_ILI9163C_registers.h:6:22: error: expected '}' before numeric constant
 #define CMD_NOP      0x00//Non operation
                      ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:38:5: note: in expansion of macro 'CMD_NOP'
     CMD_NOP = 0x00,
     ^
/Users/JimD/Documents/Arduino/libraries/TFT_ILI9163C-master/_settings/TFT_ILI9163C_registers.h:6:22: error: expected unqualified-id before numeric constant
 #define CMD_NOP      0x00//Non operation
                      ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:38:5: note: in expansion of macro 'CMD_NOP'
     CMD_NOP = 0x00,
     ^
In file included from /Users/JimD/Documents/Arduino/RewinderTS/RewinderTS.ino:14:0:
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:65:12: error: expected unqualified-id before ')' token
   MLX90393();
            ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:116:1: error: expected unqualified-id before 'private'
 private:
 ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:138:3: error: expected unqualified-id before 'private'
   private:
   ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:141:1: error: expected declaration before '}' token
 };
 ^
exit status 1
Error compiling for board Adafruit Feather M0.

tedyapo, 2019-07-17:
The other library should not pollute the global namespace with #defines like CMD_NOP. If they used scoped enums like this one does, this kind of error would not happen. I see you have filed a bug with the other library - you can add the fact that they should not use such #defines if you like.

Thanks for your interest in this library and for reporting the issue. I'm going to close it because it's a problem with the other library.

tedyapo, 2019-07-17:
you can try re-arranging the order of the includes for the library headers, but I don't remember if arduino's "improvements" to header file processing will interfere with this strategy

Jim-bee, 2019-07-17:
Hi Ted,

Thank you for your reply.  I try this and make your recommendation to them.  Thanks again for your help.

Best Regards,
Jim

tedyapo, 2019-07-18:
it's possible to clean up the #defines a little if that style is still preferred. If you add a scoping prefix, you're unlikely to get such collisions. For instance:

`#define TFT_ILI9163C_CMD_NOP 0x00//Non operation`

is unlikely to collide with anything in other libraries. But, I still prefer the enums which can be scoped within the class.

Jim-bee, 2019-07-18:
Hi Ted,

Thanks I will give that a try tomorrow.

Thanks Again,
Jim

tedyapo, 2020-08-08:
i am tired of seeing this open

closed 2020-08-08 by tedyapo