tedyapo/arduino-MLX90393 #45: Help with MLX90393 product from Adafruit/ Burst Mode

jeg100, 2019-10-23:
Hello,

I'll preface this by saying I'm very new to Arduino and I2C communication, so I apologize if I'm missing something obvious. 

I have a MLX90393 sensor bought from Adafruit Industries last spring. I was able to get measurements using their example code and library (which only supports Single Measurement Mode). Using Single Measurement Mode in a loop with no delay only gives about a 21 Hz sampling rate, but I see in the data sheet that over 400 Hz is possible when the sensor is set to Burst Mode. I tried to modify their library to set the sensor to Burst Mode, but this caused the initialization of the sensor through their begin() function to fail, and after not receiving answers on their forums I looked for other libraries with burst mode support and found this.

My problem right now is that I can't even get Single Measurement mode to work with this library. I've been trying to test/debug using the following code:

```
#include <Wire.h>
#include <Arduino.h>
#include <MLX90393.h>

MLX90393 sensor = MLX90393();

void setup() {
  Serial.begin(9600);

  while (!Serial) {
    delay(10);
  }

  Wire.begin();

  if (sensor.begin() == 0)
  {
    Serial.println("Sensor connected successfully.");
  }
  else
  {
    Serial.println("Error occured.");
  }
}

void loop() {
  MLX90393::txyz data;

  uint8_t status1 = sensor.readData(data);
  Serial.println(status1, 4);
  Serial.println(data.t, 4);
  Serial.println(data.x, 4);
  Serial.println(data.y, 4);
  Serial.println(data.z, 4);
  delay(1000);
}
```

I have no problems getting begin() to be successful (it always returns 0), but the displayed data is very obviously incorrect. For example, the temperature value is 459.792, and the magnetic field values are -0.15 or -0.24, and they don't change when I put a magnet near the sensor. I checked to see where there may be an error occuring in readData, and found that checkStatus(status1) (status for startMeasurement() returns 0 in the serial monitor, but checkStatus(status2) returns 3333 (status2 itself is 302). checkStatus(status1) | checkStatus(status2) also returns 3333.

I think the strangest thing is that there was one time where I actually got believable measurements (that responded to a magnet placed near sensor), but when I reuploaded the sketch it reverted back to the previous issue.

My questions would be, do my issues stem from using the Adafruit product, and if so what might I need to modify in the library to get it to work with this product? Also, is it even worth trying to get Burst Mode working with this sensor (I saw elsewhere that readings in Burst Mode often fail)?

Thank you for any help with this.

tedyapo, 2019-10-30:
A couple of questions:

1. Can you provide a link to the Adafruit hardware you are using?

2. What Arduino hardware are you using?

3. Can you explain how the hardware is connected together?

An intermittent condition like you think you saw could be improper I2C pullup resistors. Just a wild guess.

I can't address the specific burst mode questions; never used it.

jeg100, 2019-10-30:
Thanks for getting back to me. To answer your questions:

1. Here's a link to the Adafruit MLX90393 product: [https://learn.adafruit.com/mlx90393-wide-range-3-axis-magnetometer?view=all](). 

2. I've tried this with both an Arduino Nano and an Elegoo Uno R3.

3. In both cases, I'm using the A4/A5 pins and 5V pin on the board to connect to the SDA/SCL pins and Vin respectively on the sensor (along with connecting ground pins). I haven't been using the interrupt pin.

I also haven't been using my own pull-up resistors since the guide for the product says you should be able to directly connect the 5V pin of 5V microcontrollers to the sensor, and the SDA/SCL pins have their own internal pull-up resistors. The interrupt pin does need to be level shifted to 3.3V, but as I said I haven't used it yet. I haven't really had any problems with the readings with either of the microcontroller boards when using Adafruit's library and their example sketch, although it does fail if I try to upload the sketch after trying my above sketch for this library without unplugging/plugging the USB first.

I'm still trying to figure out why checkStatus2 (related to readMeasurement()) is returning the error I mentioned. Unless I'm misunderstanding something it's not outputting STATUS_ERROR. The time I did get the successful measurements checkStatus2 was 0 if I remember correctly.

j980m842, 2020-07-31:
Hey Ted,

I have had similar issues with the library but I have maybe isolated the issues a bit more. I have no trouble reading from the mlx 90393 with my code, but if I add a serial.print to the void loop I get the same symptoms that jeg100 is mentioning. Comment out the single line of code and reupload and it goes right back to correctly reading the data from the magnetometer. Does this help clear up what could be going on?

tedyapo, 2020-08-08:
i ordered one of the adafruit boards to see what is going on. it may take a while to get here, but i can finally get to the bottom of the problems people have reported to me about this board.

also, can you provide a full example of a program that shows the erroneous behavior, like in a gist?

and what arduino board are you using?

j980m842, 2020-08-08:
Hey Ted,

Firstly, I really appreciate you getting back to me and wanting to help out!

I am using Adafruit’s breakout board and I have the same issues no matter what Arduino board I am running, I have tested it with at least two different brands of Arduino UNO and with an M5Stick-C and I have the same issues. I’ve attached the simple code I’m running and you will see in the loop I have a serial print “channel 1: “ commented out. If I run the script with that commented out it works fine, if that serial print is uncommented it won’t work. In my testing so far it looks like it doesn’t matter where that serial print is, even if it is behind an if statement condition that is never reached. Also, this is just the easiest example of what trips it up. For my project I am trying to run 6 MLXs through a multiplexor and switching the channel on the multiplexor causes the same issue. I know it isn’t the multiplexor because I have run it successfully with two other types of magnetometers. If you want code for that too I could run to my lab because I don’t have it one with that issue handy.

I’m not sure if I mentioned it previously but I have also tried using Adafruit’s library for the MLX and it works perfectly, I don’t have any issues like that. However, they don’t have functions for setting OSR, digital filter, or resolution which makes the magnetometer almost useless… I just say that in case it might help you troubleshoot if you can look at the documentation for their libraries knowing it works.

Did it say how long it may take your Adafruit MLX breakout board to come in? I could ship you one if you are willing to share your address since I have several of them. Would this be helpful? Please let me know if there is anything else you need or any other information that would be helpful.

I’m a physiologist with relatively little programming experience so It is difficult for me to fix issues like this if libraries aren’t working properly and it can keep my projects from moving forward, so again I really appreciate your generosity in helping out with this!

Thank you,

Jonathan D. Miller

 
#include <Wire.h>
#include <MLX90393.h>

MLX90393 mlx;
MLX90393::txyz data; //Create a structure, called data, of four floats (t, x, y, and z)

int GAIN = 7;
int RES_X = 0;
int RES_Y = 0;
int RES_Z = 0;
int OSR = 2;
int DIG_FILT = 7;

  Wire.begin();
  
  byte status = mlx.begin(); //iic jumpers set
  Serial.println(status);

  Serial.print("Start status: 0x");
  if(status < 0x10) Serial.print("0");
  Serial.println(status, BIN);
  
  mlx.setGainSel(GAIN);
  mlx.setResolution(RES_X, RES_Y, RES_Z); //x, y, z
  mlx.setOverSampling(OSR);
  mlx.setDigitalFiltering(DIG_FILT);
}

void loop()
{
  mlx.readData(data);
  //Serial.println("channel1:  ");
  Serial.println(String(millis())+ ","+ data.x+","+data.y+","+data.z);
  delay(2);
}

tedyapo, 2020-08-25:
I believe the original issue was fixed by 4b044b532be969a4464e58fa488459d97248596d. Please pull latest and verify.

I tried @jeg100 's code from https://github.com/tedyapo/arduino-MLX90393/issues/45#issue-511402221, and it fails before that commit, but works afterwards

it remains to be seen if @j980m842 's issue is fixed by the same commit. I'm closing this issue since @j980m842 has opened another one.

closed 2020-08-25 by tedyapo

Anomis28, 2023-11-17:
Hi, I have an application similar to the one of @j980m842 and therefore I need to read data in bust mode. From your previous comment, I did not understand perfectly the steps needed to get reliable data since there were multiple issues opened on the topic. Can someone resume it? Thank you