tedyapo/arduino-MLX90393 #46: linear position

jschoch, 2019-11-14:
Hey Ted,

I have been dabbling with the ALS31313 and rummaged through a drawer and saw one of these MLX90303 sensors.  I have been toying with trying to figure out how to read linear position.  For the ALS31313 I used some code called "anaQuad" to read ~2mm pitch fridge magnet (and 1mm pole pitch strip) by looking at the Z strength and the X strength which seem 90 degrees out of phase.   I'm able to get about 4 ticks per mm for the fridge magnet and 8 on the "high resolution" strip.  Seems like I should be able to do better.   Since I don't know who else to ask, wondering if you have any ideas on how this should work with either the ALS or MLX sensors?  I also have a AS5311 which is great because it outputs an incremental signal but they are expensive and hard to source.  I'm guessing the angles are better to use but my math is quite weak.

Feel free to close right away if you don't have any ideas.

pvint, 2020-02-06:
Hey Jesse,

Can't directly help you, but I'm considering the ALS31313 after a disaster with the TLE493D (it's a real pain...). Just curious - what version of the ALS31313 did you test? A quick look at the datasheet shows me that the ALS31313KLEATR-JOY has a range of +/- 200 mT, except for the Z-axis which is +/- 800 mT. I'm looking at an application using magnets about the same size, so I'm curious as to which version of the chip you used.

Cheers
Paul

jschoch, 2020-02-07:
i tested the ALS31313KLEATR-1000 and the ALS31313KLEATR-JOY.  the joy was definitely not as sensitive as the -1000.   I was able to get about .02 mm resolution, I made a video about it here:  https://youtu.be/lxywqETVLhE

I bought some TMR sensors but I have not had time to set them up yet.

pvint, 2020-02-07:
Haha! I had just watched your video before posting here... I'll follow up there after having another look.

Thanks

closed 2020-08-08 by tedyapo