tedyapo/arduino-MLX90393 #48: strange issues with serial.print and other things

j980m842, 2020-07-31:
Hello,

I am using an Arduino Uno (IDE version 1.8.12) with an MLX90393 magnetometer. My issue is that I can read from the magnetometer just fine if pretty much ONLY do the readData command and print it such as shown below:

void loop()
{
mlx.readData(data);
Serial.println(String(millis())+ ","+ data.x+","+data.y+","+data.z);
delay(500);
}

However, If I add other things to the void loop such as also serial printing the channel I am reading from like this:

void loop()
{
mlx.readData(data);
Serial.println("channel1: ");
Serial.println(String(millis())+ ","+ data.x+","+data.y+","+data.z);
delay(500);
}

It not longer reads my data and I get something like "0.75, 0.75, 1.58" for my three data points each time it samples and the values don't change if I wave a magnet over the magnetometer so I know it isn't actually reading anything. This is just an example, there are a host of other strange things that seem to keep the magnetometers from reading when they are compiled on the board. Also, I have noticed that every time I start the serial monitor, the first data point I get is a nonsense data point such as "0.75, 0.75, 1.58" and then after that it begins giving me accurate X, Y, Z values. As an addition, I am also trying to read several MLX90393s using a TCA9548 and I'm having similar problems when using that because even changing the channel to read from will keep all of the magnetometers from reading data and again they just give nonsense numbers. Any suggestions what could be causing this?

tedyapo, 2020-08-08:
questions:

1. what MLX90393 board are you using? A commercial one, or one of your own design?
2. Can you provide a complete code example (not just the main loop) that illustrates the problem?

tedyapo, 2020-08-25:
I can't reproduce your example exactly here, because the code you supplied doesn't compile -- there's an error on the Serial.println(String(millis()...

However, I was able to see the code fail if I fixed that error and pulled an older version of the code. I believe this error was fixed by 4b044b532be969a4464e58fa488459d97248596d. If I use the code after that fix, I have not been able to reproduce your error.

Please pull the latest and try your code again.